espressif/esp-adf #1472: algorithm example not working on lyrat mini v1.2

keeturan, 2025-06-20:
## Environment

- Audio development kit:      [ESP32-LyraT-Mini|]
- Audio kit version (fESP32-LyraT-Mini]
- [Required] Module or chip used:  [ESP32-WROVER-E]
- [Required] IDF version: v5.4
- [Required] ADF version: v2.7-105-g4200c64d
- Build system:         [CMake]
- [Required] Running log:

`--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6592
load:0x40078000,len:16664
load:0x40080400,len:4
--- 0x40080400: _init at ??:?

ho 8 tail 4 room 4
load:0x40080404,len:4268
entry 0x40080658
I (31) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (31) boot: compile time Jun 20 2025 08:12:10
I (31) boot: Multicore bootloader
I (33) boot: chip revision: v3.1
I (36) qio_mode: Enabling default flash chip QIO
I (40) boot.esp32: SPI Speed      : 80MHz
I (44) boot.esp32: SPI Mode       : QIO
I (47) boot.esp32: SPI Flash Size : 8MB
I (51) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00200000
I (84) boot:  3 model            Unknown data     01 82 00210000 00200000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e314h (123668) map
I (133) esp_image: segment 1: paddr=0002e33c vaddr=3ff80000 size=0001ch (    28) load
I (133) esp_image: segment 2: paddr=0002e360 vaddr=3ffb0000 size=01cb8h (  7352) load
I (139) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=43e04h (278020) map
I (215) esp_image: segment 4: paddr=00073e2c vaddr=3ffb1cb8 size=020d0h (  8400) load
I (218) esp_image: segment 5: paddr=00075f04 vaddr=40080000 size=13534h ( 79156) load
I (252) boot: Loaded app from partition at offset 0x10000
I (252) boot: Disabling RNG early entropy source...
I (262) quad_psram: This chip is ESP32-D0WD
I (263) esp_psram: Found 8MB PSRAM device
I (263) esp_psram: Speed: 80MHz
I (263) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (270) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (278) cpu_start: Multicore app
I (795) esp_psram: SPI SRAM memory test OK
I (803) cpu_start: Pro cpu start user code
I (803) cpu_start: cpu freq: 240000000 Hz
I (803) app_init: Application information:
I (803) app_init: Project name:     algorithm_examples
I (808) app_init: App version:      1
I (811) app_init: Compile time:     Jun 20 2025 08:15:10
I (816) app_init: ELF file SHA256:  5b7789baa...
I (820) app_init: ESP-IDF:          v5.4-dirty
I (825) efuse_init: Min chip rev:     v0.0
I (828) efuse_init: Max chip rev:     v3.99 
I (832) efuse_init: Chip rev:         v3.1
I (836) heap_init: Initializing. RAM available for dynamic allocation:
I (843) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (848) heap_init: At 3FFB4A28 len 0002B5D8 (173 KiB): DRAM
[... 246 lines cut ...]

    }

    ESP_LOGI(TAG, "[8.0] Stop audio_pipeline");

    audio_pipeline_stop(pipeline_rec);
    audio_pipeline_wait_for_stop(pipeline_rec);
    audio_pipeline_deinit(pipeline_rec);

    audio_element_deinit(i2s_stream_reader);
    audio_element_deinit(i2s_stream_writter);

    /* Terminate the pipeline before removing the listener */

    /* Stop all periph before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    esp_periph_set_destroy(set);
}

```

keeturan, 2025-06-25:
Hi all, Thank you for your answer. 

The speaker's output is the voice signal output by the algorithm. I'm not using microphone or referance input directly. The algorithm is used as in the example, I don't understand the difference. The dac output reference signal comes from the i2s left channel and the right channel is the microphone. As in the example. Isn't the aec algorithm used this way? Please correct me if I'm wrong.

This is the point I particularly wanted to test in this example. I tried the same algorithm with the ring buffered version on my special development card other than Lyrat and I could see the difference of the algorithm on the echo. The problem is that my own development card does not have 2 ADC inputs, I applied this example with the reference buffer. I want to test the AEC performance with synchronous i2s data without using the ring buffer on Lyrat.

Technically, can I import the aec algorithm into pipeline_a2dp_sink_and_hfp application using esp32?

When I try to add it using the aec ring buffered version in the hf application, the aec algorithm has no effect. In the hf application, the microphone on the esp side only delays the sound by 4 seconds.

keeturan, 2025-07-10:
Hi all,

It works. The AEC algorithm works successfully. Thank you for your answer.

After seeing the successful operation of AEC, I added the AEC algorithm to the pipe_a2dp_sink_and_hfp application in the code below.

I have only one problem. In the HF application, the sound from the other side comes instantly. But the voice data I receive from the algorithm output reaches the other side after 4 seconds. I don't see anything in the code below that would cause a 4 second delay.

Is this related to processor performance or algorithm performance? What could be the reason?

`/* Play music from Bluetooth device

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <inttypes.h>
#include "nvs_flash.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include "esp_peripherals.h"
#include "periph_touch.h"
#include "periph_adc_button.h"
#include "periph_button.h"
#include "esp_bt_defs.h"
#include "esp_gap_bt_api.h"
#include "esp_hf_client_api.h"

#include "audio_element.h"
#include "audio_pipeline.h"
#include "audio_event_iface.h"
#include "audio_mem.h"

#include "i2s_stream.h"
#include "board.h"
#include "bluetooth_service.h"
#include "filter_resample.h"
#include "raw_stream.h"

#if (CONFIG_ESP_LYRATD_MSC_V2_1_BOARD || CONFIG_ESP_LYRATD_MSC_V2_2_BOARD)
#include "filter_resample.h"
#endif
#include "bt_keycontrol.h"

#include "audio_idf_version.h"

#if (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 0, 0))
#define HFP_RESAMPLE_RATE 16000
#else
#define HFP_RESAMPLE_RATE 8000
#endif

static const char *TAG = "BLUETOOTH_EXAMPLE";
static const char *BT_HF_TAG = "BT_HF";

static audio_element_handle_t  bt_stream_reader, i2s_stream_writer, i2s_stream_reader;
static audio_pipeline_handle_t pipeline_d, pipeline_e;
static bool is_get_hfp = true;
ringbuf_handle_t ringbuf_algo_out;

#include "driver/i2s.h"
#include "audio_sys.h"
#include "algorithm_stream.h"
#define I2S_CHANNELS        I2S_CHANNEL_FMT_RIGHT_LEFT
#define I2S_BITS            CODEC_ADC_BITS_PER_SAMPLE
[... 541 lines cut ...]
    audio_pipeline_unregister(pipeline_d, filter_d);
    audio_pipeline_unregister(pipeline_e, filter_e);
#endif
    /* Terminate the pipeline before removing the listener */
    audio_pipeline_remove_listener(pipeline_d);

    /* Stop all peripherals before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    /* Release all resources */
    audio_pipeline_deinit(pipeline_d);
    audio_element_deinit(bt_stream_reader);
    audio_element_deinit(i2s_stream_writer);
    audio_element_deinit(i2s_stream_reader);
#if (CONFIG_ESP_LYRATD_MSC_V2_1_BOARD || CONFIG_ESP_LYRATD_MSC_V2_2_BOARD)
    audio_element_deinit(filter_d);
    audio_element_deinit(filter_e);
#endif
    esp_periph_set_destroy(set);
    bluetooth_service_destroy();
}`

jason-mao, 2026-03-04:
Hi @keeturan , 
ESP32 uses AEC for HFP, and it works well at an 8 kHz sample rate. We have tested it at 8 kHz, and it performs well.