espressif/esp-adf #1495: coze_ws_app 执行函数esp_err_t audio_playback_close(void)无法同过

Freak-luo, 2025-07-10:
在调用了audio_playback_close（）的函数后发现，audio_playback_stop中的esp_audio_simple_player_stop一直无法停止管道，并报错**E (77592) ESP_GMF_TASK: Stop timeout,[TSK_0x3fce4cf8,0x3fce4cf8]** 这样会导致audio_playback_close(void)函数中 esp_err_t err = esp_audio_simple_player_destroy(audio_playback.player);的这一行函数中 代码执行到源码  xEventGroupClearBits(player->wait_event, ASP_PIPELINE_ERROR_BIT | ASP_PIPELINE_STOPPED_BIT | ASP_PIPELINE_FINISHED_BIT);这里的时候会崩溃。导致程序重启。是清除管道需要有特定的循序吗？还是这个代码没完善。

jason-mao, 2025-07-17:
帮提供一下gmf_core 的版本，我们在最新 GMF core V0.7 上修复了一个timeout问题。建议你用最新的coze app再测试一下。

Freak-luo, 2025-07-17:
**gmf_core 的版本就是0.7的版本  IDF的版本是v5.4.1  _ADF版本是master**_
调用audio_manager_deinit退出coze聊天代码
esp_err_t coze_chat_app_stop(void)
{
    esp_gmf_err_t ret = audio_manager_deinit();
    ESP_LOGI(TAG, "audio_manager_deinit RET ==== %s", esp_err_to_name(ret));
    if (coze_chat.chat)
    {
        esp_coze_chat_stop(coze_chat.chat);
        ESP_LOGI(TAG, "esp_coze_chat_stop RET ==== %s", esp_err_to_name(ret));
        esp_coze_chat_deinit(coze_chat.chat);
        ESP_LOGI(TAG, "esp_coze_chat_deinit RET ==== %s", esp_err_to_name(ret));
    }
    return ESP_OK;
}
调用audio_manager_deinit()方法关闭audio_palyback日志
I (32958) COZE_CHAT_APP: vad end
I (32959) AUDIO_PROCESSOR: VAD_END
I (35327) COZE_CHAT_APP: vad start
I (35328) AUDIO_PROCESSOR: VAD_START
I (38342) COZE_CHAT_APP: vad end
I (38342) AUDIO_PROCESSOR: VAD_END
I (38537) ESP_COZE_CHAT: Terminal :{"id":"18f39451-843b-6182-bcec-7150a5dfca46","event_type":"conversation.chat.cancel"}
I (39042) AUDIO_PROCESSOR: Deinitializing audio manager...
I (39043) AUDIO_PROCESSOR: Closing audio recorder...
I (39075) AFE_MANAGER: AFE manager suspend 1
I (39075) ESP_GMF_TASK: One times job is complete, del[wk:0x3c320cc8, ctx:0x3c316acc, label:ai_afe_close]
I (39077) ESP_GMF_TASK: One times job is complete, del[wk:0x3c345dfc, ctx:0x3c282edc, label:aud_enc_close]
I (39086) ESP_GMF_TASK: Waiting to run... [tsk:audio_recorder_task-0x3fce45f4, wk:0x0, run:0]
I (39095) ESP_GMF_TASK: Waiting to run... [tsk:audio_recorder_task-0x3fce45f4, wk:0x0, run:0]
I (39103) AFE: exit
I (39105) AFE_MANAGER: AFE manager destroy
I (39108) AUDIO_PROCESSOR: Audio recorder closed successfully
I (39114) AUDIO_PROCESSOR: Closing audio playback...
I (39119) AUDIO_PROCESSOR: Stopping audio playback...
E (42123) ESP_GMF_TASK: Stop timeout,[audio_playback_task,0x3fce4e70]
W (42123) AUDIO_PROCESSOR: esp_gmf_task_get_state...3
E (42123) AUDIO_PROCESSOR: audio_processor.c:1179 (audio_playback_stop): Failed to stop playback pipeline

单独调用audio_playback_close来关闭playback代码
esp_err_t coze_chat_app_stop(void)
{
    esp_gmf_err_t ret = esp_gmf_oal_thread_delete(coze_chat.read_thread);
    // esp_gmf_err_t ret = audio_manager_deinit();
    ret = audio_recorder_close();
    ESP_LOGI(TAG, "audio_recorder_close RET ==== %s", esp_err_to_name(ret));
    if (coze_chat.chat)
    {
        esp_coze_chat_stop(coze_chat.chat);
        ESP_LOGI(TAG, "esp_coze_chat_stop RET ==== %s", esp_err_to_name(ret));
        esp_coze_chat_deinit(coze_chat.chat);
        ESP_LOGI(TAG, "esp_coze_chat_deinit RET ==== %s", esp_err_to_name(ret));
    }
    ret = audio_playback_close();
    ESP_LOGI(TAG, "audio_playback_close RET ==== %s", esp_err_to_name(ret));
    return ESP_OK;
}

单独调用audio_playback_close来关闭playback日志
I (51542) AUDIO_PROCESSOR: Closing audio recorder...
I (51554) AFE_MANAGER: AFE manager suspend 1
I (51555) ESP_GMF_TASK: One times job is complete, del[wk:0x3c3403a8, ctx:0x3c282f88, label:ai_afe_close]
I (51557) ESP_GMF_TASK: One times job is complete, del[wk:0x3c280468, ctx:0x3c316acc, label:aud_enc_close]
I (51566) ESP_GMF_TASK: Waiting to run... [tsk:audio_recorder_task-0x3fcdd11c, wk:0x0, run:0]
I (51574) ESP_GMF_TASK: Waiting to run... [tsk:audio_recorder_task-0x3fcdd11c, wk:0x0, run:0]
I (51583) AFE: exit
I (51584) AFE_MANAGER: AFE manager destroy
I (51588) AUDIO_PROCESSOR: Audio recorder closed successfully
I (51594) COZE_CHAT_APP: audio_recorder_close RET ==== ESP_OK
I (51599) ESP_COZE_CHAT: coze_data_pull_task exit
I (51599) ESP_COZE_CHAT: coze_data_push_task exit
I (52576) COZE_CHAT_APP: esp_coze_chat_stop RET ==== ESP_OK
I (52576) COZE_CHAT_APP: esp_coze_chat_deinit RET ==== ESP_OK
I (52576) AUDIO_PROCESSOR: Closing audio playback...
I (52580) AUDIO_PROCESSOR: Stopping audio playback...
E (55585) ESP_GMF_TASK: Stop timeout,[audio_playback_task,0x3fcdd9bc]
W (55585) AUDIO_PROCESSOR: esp_gmf_task_get_state...3
E (55585) AUDIO_PROCESSOR: audio_processor.c:1179 (audio_playback_stop): Failed to stop playback pipeline

shootao, 2025-07-17:
Hi @Freak-luo 感谢反馈。可以先修改以下两点试一下
1. 修改 audio_processor.c, 如下
```c
diff --git a/examples/ai_agent/coze_ws_app/main/audio_processor.c b/examples/ai_agent/coze_ws_app/main/audio_processor.c
index 4469f9f47..0151a77d1 100644
--- a/examples/ai_agent/coze_ws_app/main/audio_processor.c
+++ b/examples/ai_agent/coze_ws_app/main/audio_processor.c
@@ -842,7 +842,10 @@ static int playback_inport_acquire_read(void *handle, esp_gmf_data_bus_block_t *
     _blk.buf_length = blk->buf_length;
     int ret = esp_gmf_db_acquire_read(audio_playback.fifo, &_blk, wanted_size,
                                         block_ticks);
-    if (ret < 0) {
+    if (ret == ESP_GMF_IO_ABORT) {
+        ESP_LOGI(TAG, "Fifo acquire read aborted");
+        return ESP_GMF_IO_ABORT;
+    } else if (ret < 0) {
         ESP_LOGE(TAG, "Fifo acquire read failed (0x%x)", ret);
         return ESP_FAIL;
     }
@@ -851,7 +854,10 @@ static int playback_inport_acquire_read(void *handle, esp_gmf_data_bus_block_t *
 #else
     int ret = esp_gmf_db_acquire_read(audio_playback.fifo, &_blk, wanted_size,
                                         block_ticks);
-    if (ret < 0) {
+    if (ret == ESP_GMF_IO_ABORT) {
+        ESP_LOGI(TAG, "Fifo acquire read aborted");
+        return ESP_GMF_IO_ABORT;
+    } else if (ret < 0) {
         ESP_LOGE(TAG, "Fifo acquire read failed (0x%x)", ret);
         return ESP_FAIL;
     }
@@ -973,13 +979,11 @@ static esp_err_t playback_configure_converters(void)
     if (rate_cvt_el) {
         esp_gmf_rate_cvt_set_dest_rate(rate_cvt_el, BOARD_SAMPLE_RATE);
     }
-
     esp_gmf_element_handle_t ch_cvt_el = NULL;
     esp_gmf_pipeline_get_el_by_name(audio_playback.pipe, "aud_ch_cvt", &ch_cvt_el);
     if (ch_cvt_el) {
         esp_gmf_ch_cvt_set_dest_channel(ch_cvt_el, BOARD_CHANNELS);
     }
-
     return ESP_OK;
 }

@@ -1035,6 +1039,13 @@ static esp_err_t playback_configure_pipeline(void)
     return ESP_OK;
 }

+static esp_err_t playback_prev_stop_cb(void *ctx)
+{
+    ESP_LOGI(TAG, "Previous stop callback");
+    esp_gmf_db_abort(audio_playback.fifo);
+    return ESP_OK;
+}
+
 esp_err_t audio_playback_open(void)
```
2. 将[esp_gmf_audio_dec.c](https://github.com/espressif/esp-gmf/blob/main/elements/gmf_audio/esp_gmf_audio_dec.c#L251) 改为 `ESP_GMF_PORT_ACQUIRE_IN_CHECK(TAG, load_ret, out_len, {out_len = ESP_GMF_JOB_ERR_FAIL; goto __aud_proc_release;});`