espressif/esp-adf #1510: About OPUSDECODE->I2S WRTIE, AEC->OPUSENCODE

EternityFOR, 2025-07-26:
具体配置如下
```
#define I2S_CHANNEL_NUM            1
#define I2S_CHANNELS               I2S_CHANNEL_TYPE_ONLY_LEFT
#define I2S_SAMPLE_RATE            8000
#define I2S_BITS                   CODEC_ADC_BITS_PER_SAMPLE

#define PCM_DURATION               20 / 1000
#define PCM_SAMPLE_RATE            16000
#define PCM_BUFFER_SIZE            I2S_SAMPLE_RATE * PCM_DURATION * I2S_CHANNEL_NUM * sizeof(int16_t) // 640 bytes

#define OPUS_CHANNEL_NUM           2
#define OPUS_ENCODE_DURATION       20 / 1000
#define OPUS_ENCODE_SAMPLE_RATE    16000
#define OPUS_ENCODE_BUFFER_SIZE    OPUS_ENCODE_SAMPLE_RATE * OPUS_ENCODE_DURATION * OPUS_CHANNEL_NUM * sizeof(int16_t) // 1280 bytes

#define OPUS_DECODE_DURATION       120 / 1000
#define OPUS_DECODE_SAMPLE_RATE    16000
#define OPUS_DECODE_BUFFER_SIZE    OPUS_DECODE_SAMPLE_RATE * OPUS_DECODE_DURATION * OPUS_CHANNEL_NUM * sizeof(int16_t) // 7680 bytes
#define OPUS_DECODE_FRAME_SIZE     OPUS_DECODE_SAMPLE_RATE / 400
static esp_err_t init_server_sub_pipeline(void)
{
    ESP_LOGI(TAG, "Initializing Server Subscriber pipeline");
    //***  Server subscriber pipeline Initialization  ***/
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    server_sub_pipeline = audio_pipeline_init(&pipeline_cfg);
    if (!server_sub_pipeline) {
        ESP_LOGE(TAG, "Failed to create server pipeline");
        return ESP_FAIL;
    }
    
    //***   INPUT: Create RAW stream reader for server audio  ***/
    raw_opus_dec_cfg_t dec_cfg = RAW_OPUS_DEC_CONFIG_DEFAULT();
    dec_cfg.channels = 2;
    dec_cfg.sample_rate = OPUS_DECODE_SAMPLE_RATE;
    dec_cfg.out_rb_size = OPUS_DECODE_BUFFER_SIZE;
    dec_cfg.dec_frame_size = OPUS_DECODE_FRAME_SIZE;
    dec_cfg.task_core = 1;
    dec_cfg.task_prio = 15;
    dec_cfg.task_stack = 16384;
    audio_element_handle_t server_opus_decoder = raw_opus_decoder_init(&dec_cfg);
    audio_element_set_read_cb(server_opus_decoder, server_sub_cb, NULL);

    //***   OUTPUT: Create Resample Filter for server audio  ***/
    rsp_filter_cfg_t rsp_cfg_w = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_cfg_w.src_rate  = OPUS_DECODE_SAMPLE_RATE;
    rsp_cfg_w.src_ch    = 2;   
    rsp_cfg_w.dest_rate = I2S_SAMPLE_RATE;
    rsp_cfg_w.dest_ch   = 1;
    rsp_cfg_w.out_rb_size = OPUS_DECODE_BUFFER_SIZE;
    rsp_cfg_w.complexity = 1;
    rsp_cfg_w.task_core = 1;
    rsp_cfg_w.task_prio = 15;
    rsp_cfg_w.task_stack = 16384;
    audio_element_handle_t server_resample_filter = rsp_filter_init(&rsp_cfg_w);
    audio_element_set_write_cb(server_resample_filter, i2s_write_cb, NULL);
    audio_element_set_output_timeout(server_resample_filter, portMAX_DELAY);

    audio_pipeline_register(server_sub_pipeline, server_opus_decoder, "server_opus_decoder");
    audio_pipeline_register(server_sub_pipeline, server_resample_filter, "server_resample");
    
    ESP_LOGI(TAG, "Link it together [flash]-->raw_opus_decoder-->filter-->[codec_chip]");
    const char *link_tag[2] = {"server_opus_decoder", "server_resample"};
    audio_pipeline_link(server_sub_pipeline, &link_tag[0], 2);
    audio_pipeline_run(server_sub_pipeline);
    ESP_LOGI(TAG, "Server Subscriber pipeline initialized successfully");
    return ESP_OK;
}
static esp_err_t init_server_pub_pipeline(void)
{
    //***  Server publisher pipeline Initialization  ***/
    ESP_LOGI(TAG, "Initializing Server Publisher pipeline");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    server_pub_pipeline = audio_pipeline_init(&pipeline_cfg);
    if (!server_pub_pipeline) {
        ESP_LOGE(TAG, "Failed to create Server Publisher pipeline");
        return ESP_FAIL;
    }

    //***   INPUT: AEC Element Initialization  ***/
    aec_stream_cfg_t aec_cfg = AEC_STREAM_CFG_DEFAULT();
    aec_cfg.input_format = "RM";
    // aec_cfg.mode = AFE_MODE_HIGH_PERF;
    // aec_cfg.type = AFE_TYPE_VC;
    aec_cfg.task_core = 0;
    aec_cfg.task_prio = 14;
    aec_cfg.task_stack = 16384 * 50;
    audio_element_handle_t aec_element = aec_stream_init(&aec_cfg);
    audio_element_set_read_cb(aec_element, i2s_read_cb, NULL);

    //***   OUTPUT: Create RAW stream writer for server audio  ***/
    raw_opus_enc_config_t enc_cfg = RAW_OPUS_ENC_CONFIG_DEFAULT();
    enc_cfg.channel = I2S_CHANNEL_NUM;
    enc_cfg.out_rb_size = OPUS_ENCODE_BUFFER_SIZE;
    enc_cfg.bitrate = OPUS_ENCODE_SAMPLE_RATE;
    enc_cfg.complexity = 1;
    enc_cfg.task_core = 0;
    enc_cfg.task_prio = 14;
    enc_cfg.task_stack = 16384 * 50;
    enc_cfg.application_mode = RAW_OPUS_ENC_APPLICATION_AUDIO;
    audio_element_handle_t server_opus_encoder = raw_opus_encoder_init(&enc_cfg);
    audio_element_set_write_cb(server_opus_encoder, server_pub_cb, NULL);
    audio_element_set_output_timeout(server_opus_encoder, portMAX_DELAY);

    audio_pipeline_register(server_pub_pipeline, aec_element, "aec");
    audio_pipeline_register(server_pub_pipeline, server_opus_encoder, "server_opus_encoder");
    ESP_LOGI(TAG, "Link it together [flash]-->aec-->server_opus_encoder-->[codec_chip]");
    const char *link_tag[2] = {"aec", "server_opus_encoder"};
    audio_pipeline_link(server_pub_pipeline, &link_tag[0], 2);
    audio_pipeline_run(server_pub_pipeline);
    ESP_LOGI(TAG, "Server Publisher pipeline initialized successfully");
    return ESP_OK;
}
```
硬件配置和ESP32S3 KORVO一致，出现问题如下：
```
***ERROR*** A stack overflow in task aec has been detected.
```
这是处理速率不够快吗，很不理解我这整个流程有什么会导致这个问题。

jason-mao, 2025-07-30:
这个是 aec 这个task 的 stack size不够，你需要开大一点儿。

EternityFOR, 2025-07-30:
> 这个是 aec 这个task 的 stack size不够，你需要开大一点儿。

应该和这个无关的，我现在开很小也没问题，感觉好像和后面的处理不够及时有关，现在我的配置是这样的，还是有点问题，就是好像服务器那边听不清我在说什么，不知道怎么解决好。

jason-mao, 2025-08-13:
你的问题是`***ERROR*** A stack overflow in task aec has been detected.`？ 这个错误一般是和stack相关，或者是代码有越界可能