espressif/esp-adf #1513: esp_codec_dev_open函数求解

via1968, 2025-08-05:
为什么调用esp_codec_dev_open(codec_dev, &fs);进行初始化的时候会打印 channel:2?
还有就是为什么感觉out vol设置为100感觉声音还是比较小，和喇叭也有关吗？
使用的是esp32s3n16r8，codec为es8311 同时使用adc和dac
```
// Need install driver (i2c and i2s) firstly
	int ret = ut_i2c_init(0);
	assert(ret == 0);
	printf("i2c init ok\n");
	ret = ut_i2s_init(0);
	assert(ret == 0);
	printf("i2s init ok\n");
	// Do initialize of related interface: data_if, ctrl_if and gpio_if
	audio_codec_i2s_cfg_t i2s_cfg = {
		.rx_handle = i2s_keep[0]->rx_handle,
		.tx_handle = i2s_keep[0]->tx_handle,
	};
	const audio_codec_data_if_t *data_if = audio_codec_new_i2s_data(&i2s_cfg);
	assert(data_if != NULL);
	printf("i2s data if ok\n");
	audio_codec_i2c_cfg_t i2c_cfg = {.addr = ES8311_CODEC_DEFAULT_ADDR};
	i2c_cfg.bus_handle = i2c_bus_handle;
	const audio_codec_ctrl_if_t *out_ctrl_if = audio_codec_new_i2c_ctrl(&i2c_cfg);
	assert(out_ctrl_if != NULL);
	printf("i2c ctrl if ok\n");
	const audio_codec_gpio_if_t *gpio_if = audio_codec_new_gpio();
	assert(gpio_if != NULL);
	printf("gpio if ok\n");
	// New output codec interface
	esp_codec_dev_hw_gain_t hw_gain = {
		.pa_voltage = 5.0,
		.codec_dac_voltage = 3.3,
	};
	es8311_codec_cfg_t es8311_cfg = {
		.codec_mode = ESP_CODEC_DEV_WORK_MODE_BOTH,
		.ctrl_if = out_ctrl_if,
		.gpio_if = gpio_if,
		.pa_pin = GPIO_NUM_NC,
		.use_mclk = true,
		.digital_mic = false,
		.hw_gain = hw_gain,
		.no_dac_ref = true,
	};
	const audio_codec_if_t *out_codec_if = es8311_codec_new(&es8311_cfg);
	assert(out_codec_if != NULL);
	printf("es8311 codec if ok\n");
	// New output codec device
	esp_codec_dev_cfg_t dev_cfg = {
		.codec_if = out_codec_if,			   // codec interface from es8311_codec_new
		.data_if = data_if,					   // data interface from audio_codec_new_i2s_data
		.dev_type = ESP_CODEC_DEV_TYPE_IN_OUT, // codec support both playback and record
	};
	codec_dev = esp_codec_dev_new(&dev_cfg);
	assert(codec_dev != NULL);
	printf("codec dev ok\n");
	// Below code shows how to play
	esp_codec_dev_set_out_vol(codec_dev, 100.0);
	esp_codec_dev_sample_info_t fs = {
		.sample_rate = 16000,
		.channel = 1,
		.channel_mask = 1,
		.bits_per_sample = 16,
	};
	esp_codec_dev_open(codec_dev, &fs);
```
I (701) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (711) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:1
I (711) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (721) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:1

via1968, 2025-08-05:
```
uint8_t *raw_data2 = NULL;
int raw_size2 = 0;
void opus_encoder(void *arg)
{
    // Backup original heap size
    int heap_size = esp_get_free_heap_size();

	esp_opus_enc_config_t opus_cfg = {
		.sample_rate = ESP_AUDIO_SAMPLE_RATE_16K,
		.channel = ESP_AUDIO_MONO,
		.bits_per_sample = ESP_AUDIO_BIT16,
		.bitrate = 24000,
		.frame_duration = ESP_OPUS_ENC_FRAME_DURATION_60_MS,
		.application_mode = ESP_OPUS_ENC_APPLICATION_AUDIO,
		.complexity = 5,
		.enable_fec = false,
		.enable_dtx = false,
		.enable_vbr = false,
	};
	esp_audio_enc_handle_t encoder = NULL;
	int ret = esp_opus_enc_open(&opus_cfg, sizeof(esp_opus_enc_config_t), &encoder);
	assert(ret == 0);
	if(encoder == NULL)
	{
		printf("Opus encoder creation failed!\n");
        return;
	}
    // Get needed buffer size and prepare memory
    int pcm_size = 0, raw_size = 0;
    esp_opus_enc_get_frame_size(encoder, &pcm_size, &raw_size);
	raw_size2 = raw_size;
	printf("pcm_size: %d, raw_size: %d\n",pcm_size,raw_size);

    size_t fetch_chunksize = afe_communication->get_fetch_chunksize(afe_communication_data);
	ESP_LOGI("read_task","afe_chunksize: %d\r\n", fetch_chunksize); // 512 ->32ms
	int16_t *audio_frame_buffer = (int16_t *)calloc(1,2 * fetch_chunksize * sizeof(int16_t)); // 原始音频
	assert(audio_frame_buffer);		   // Check if audio_buffer allocation success
    uint8_t *raw_data = calloc(1,raw_size);
	afe_fetch_result_t *resu = NULL;
	esp_audio_enc_in_frame_t in_frame = {
            .buffer = (uint8_t *)audio_frame_buffer,
            .len = 16 * 60 * 2,
        };
        esp_audio_enc_out_frame_t out_frame = {
            .buffer = raw_data,
            .len = raw_size,
        };
	int offset = 0;
	while(1)
	{
		resu = afe_communication->fetch_with_delay(afe_communication_data,portMAX_DELAY);
		memcpy(audio_frame_buffer+offset, resu->data, fetch_chunksize * sizeof(int16_t));
		offset += fetch_chunksize;
		if(offset == 2 * fetch_chunksize)
		{
			offset = 0;
			esp_audio_err_t ret = esp_opus_enc_process(encoder, &in_frame, &out_frame);
			if(ret != ESP_OK)
			{
				printf("Encoding failed at frame! Error: 0x%x\n", ret);
				break;
			}
			else
			{
				raw_data2 = calloc(1,raw_size);
				memcpy(raw_data2,raw_data,raw_size);
				// printf("Encoding success at frame!\n");
			}
			// printf("\r\n");
			// for(int i=0;i<raw_size;i++)
			// {
			// 	printf("%02X ",raw_data[i]);
			// }
			// printf("\r\n");
			send_atcom(8,(uint8_t *)raw_data,raw_size);
			break;
			vTaskDelay(3);
		}
		
	}
    

    // Clear up resources
    esp_opus_enc_close(encoder);
    free(audio_frame_buffer);
    free(raw_data);
	vTaskDelete(NULL);
}

void opus_decoder(void *arg)
{
	while(raw_data2 == NULL)
	{
		vTaskDelay(1);
	}
	esp_opus_dec_cfg_t opus_cfg = {
		.sample_rate = ESP_AUDIO_SAMPLE_RATE_16K,
		.channel = ESP_AUDIO_MONO,
		.frame_duration = ESP_OPUS_DEC_FRAME_DURATION_60_MS,
		.self_delimited = true,
	};
	esp_audio_dec_handle_t decoder = NULL;
	esp_opus_dec_open(&opus_cfg, sizeof(esp_opus_dec_cfg_t), &decoder);
	esp_audio_dec_in_raw_t in_raw = {
		.buffer = raw_data2,
		.len = raw_size2,
	};
	uint8_t *frame_data = (uint8_t *)malloc(4096);
	esp_audio_dec_out_frame_t out_frame = {.buffer = frame_data, .len = 4096};
	esp_audio_dec_info_t dec_info = {0};
	while (in_raw.len)
	{
		int ret = esp_opus_dec_decode(decoder, &in_raw, &out_frame,&dec_info);
		if (ret != ESP_AUDIO_ERR_OK && ret != ESP_AUDIO_ERR_BUFF_NOT_ENOUGH)
		{
			printf("Decoding failed at frame! Error: 0x%x\n", ret);
			break;
		}
		in_raw.buffer += in_raw.consumed;
		in_raw.len -= in_raw.consumed;
		if (ret == ESP_AUDIO_ERR_BUFF_NOT_ENOUGH)
		{
			uint8_t *new_frame_data = (uint8_t *)realloc(out_frame.buffer, out_frame.needed_size);
			if (new_frame_data == NULL)
			{
				break;
			}
			out_frame.buffer = new_frame_data;
			out_frame.len = out_frame.needed_size;
		}
	}
	free(out_frame.buffer);
	esp_opus_dec_close(decoder);
	vTaskDelete(NULL);
}
```
为什么编码出来的opus解码会失败呢？求解
E (4006) ESP_OPUS_DEC: Failed to decode opus frame, error:-4.

houhaiyan, 2025-08-05:
@via1968 esp_audio_enc_get_frame_size 是输出帧buffer size的建议值， 并不是实际的编码数据。     [uint32_t encoded_bytes; /*!< It is an output parameter which means encoded data length */](https://github.com/espressif/esp-adf-libs/blob/f8baa69620b3e4cd2170098d57b1c2e7e47f9b7c/esp_audio_codec/include/encoder/esp_audio_enc.h#L63) 这个参数是实际的编码数据量输出值。 解码需要编码out_frame.encoded_bytes 赋值给 解码in_raw.len.
编码没有置上 self_delimited。 解码不能置self_delimited=true。

via1968, 2025-08-06:
好的