espressif/esp-adf #1515: VOIP 示例/库不支持自定义 SIP 域名

liangtea, 2025-08-11:
**项目路径**
examples/protocols/voip

**反馈内容**
你好，我在使用 ESP 系列（如 ESP32-S3）进行 VoIP 功能开发时，发现目前 esp-adf示例中**无法单独设置 SIP 域（realm/domain）**，导致某些场景下无法正常注册到服务器。

在 **SIP 协议** 中，`domain`（有时称为 realm）是注册认证的重要部分，通常与 SIP 服务器地址不同。
例如：

```
SIP 服务器地址（proxy/registrar）：voip.example.com
SIP 域名（realm）：hotel1.sip.com
```

当前库似乎将 `domain` 固定为与服务器相同，导致以下情况无法支持：

* 服务器部署在公网 IP，但 SIP 域名不同
* 多个虚拟域名共享同一个 IP
* 运营商或 PBX 系统要求 realm 与服务器地址不同

**期望行为**
提供 API 或配置项，允许在初始化 VoIP 时单独设置 `domain`/`realm`，并在 REGISTER 请求的 `From` / `To` / `Authorization` 头中使用该域名。

**复现步骤**

1. 使用官方 VoIP 示例
2. 无法设置realm

**技术背景**

* 根据 RFC 3261，SIP REGISTER 的认证字段 `realm` 应该由服务器挑战响应（401/407）提供，或由客户端配置。
* 目前实现中缺少显式配置，限制了与部分 SIP 服务的兼容性。
* 在多域部署（如酒店多分店 PBX）中，域名是区分租户的重要参数。

jimforr, 2025-08-15:
@liangtea 你的需求是否为指定VOIP示例启动时注册的服务器，如若是，可尝试修改以下配置
- 在 `VoIP App Configuration` > `SIP_URI` 中配置 SIP 登陆相关信息（Transport://user:password@server:port）。
  - 例如：`tcp://100:100@192.168.1.123:5060`
  - 如终端房间号为1001，默认密码为123456，你想向hotel1.sip.com注册，修改SIP_URI为（tcp://1001:123456@hotel1.sip.com:5060）

liangtea, 2025-08-15:
@jimforr 你好。需求不是你说的那样子。假设服务器的域名是：sip.example.com。
那么SIP_URI需要支持这样子：
tcp://1001@hotel1.com:123456@sip.example.com:5060
tcp://1001@hotel2.com:123456@sip.example.com:5060
以上是两个酒店，hotel1和hotel2都有1001，然后hotel1只能拨号hotel1的账号，而hotel2只能拨号hotel2的账号。

或者说可以有函数可以让我们设置realm

liangtea, 2025-08-15:
<html>
<body>
<html><head></head><body><p>我明白你的意思，你是想搞清楚 <strong>SIP realm</strong> 在连接认证中的作用，以及它和 <strong>SIP 服务器（5060 端口）</strong> 之间的区别。<br>
我用通俗但准确的方式给你拆开解释一下。</p>
<hr>
<h2>1. <strong>SIP 服务器（5060 端口）</strong></h2>
<ul>
<li>
<p><strong>是什么</strong>：<br>
SIP 服务器就是运行 SIP 协议的信令服务器（常见于 Asterisk、FreeSWITCH、Kamailio 等），它通常监听 <strong>UDP/TCP/TLS 的 5060 端口</strong> 来收发信令。</p>
</li>
<li>
<p><strong>作用</strong>：<br>
负责：</p>
<ul>
<li>
<p>注册（REGISTER）</p>
</li>
<li>
<p>呼叫建立（INVITE）</p>
</li>
<li>
<p>呼叫结束（BYE）</p>
</li>
<li>
<p>呼叫转接等</p>
</li>
</ul>
</li>
<li>
<p><strong>地址表现</strong>：</p>
<ul>
<li>
<p>可以是 IP（<code inline="">192.168.1.10:5060</code>）</p>
</li>
<li>
<p>可以是域名（<code inline="">sip.example.com:5060</code>）</p>
</li>
</ul>
</li>
</ul>
<blockquote>
<p>这个只是“通信管道”和“接线员”，不涉及你是谁、能不能用的问题。</p>
</blockquote>
<hr>
<h2>2. <strong>Realm（领域）</strong></h2>
<ul>
<li>
<p><strong>是什么</strong>：<br>
<code inline="">realm</code> 在 SIP 中是 <strong>身份认证的作用域</strong>，主要出现在 SIP Digest 认证中。<br>
它是一个字符串（通常是域名形式），告诉客户端“你要在哪个身份域下验证身份”。</p>
</li>
<li>
<p><strong>在 SIP REGISTER 时的作用</strong>：<br>
当你的 UA（用户代理）向 SIP 服务器注册时，如果服务器要求认证，它会返回类似：</p>
<pre><code>SIP/2.0 401 Unauthorized
WWW-Authenticate: Digest realm="hotelA.example.com", nonce="abc123"
</code></pre>
<ul>
<li>
<p><strong>realm</strong> 就是 <code inline="">hotelA.example.com</code></p>
</li>
<li>
<p>UA 会用这个 realm + 用户名 + 密码 + nonce 进行哈希计算，生成 <code inline="">response</code> 发回给服务器验证。</p>
</li>
</ul>
</li>
<li>
<p><strong>设计意义</strong>：</p>
<ul>
<li>
<p>可以区分不同的用户域，即使用同一台 SIP 服务器和同一个 IP，也能隔离用户组（常见于多租户系统）。</p>
</li>
<li>
<p>同一台 SIP 服务器可以配置多个 realm，例如：</p>
<pre><code>realm = hotelA.example.com
realm = hotelB.example.com
</code></pre>
<p>这样 Hotel A 的设备不会和 Hotel B 的设备互通，除非显式允许。</p>
</li>
</ul>
</li>
</ul>
<hr>
<h2>3. <strong>Realm 和 SIP 服务器的区别</strong></h2>

对比点 | SIP 服务器（5060端口） | Realm（领域）
-- | -- | --
本质 | 一个网络地址 + 端口，负责收发信令 | 一个认证作用域标识字符串
作用 | 提供 SIP 协议服务（注册、呼叫等） | 用于区分和验证用户身份
位置 | 网络层（IP:Port） | 应用层（SIP 协议认证字段）
典型值 | 192.168.1.10:5060 或 sip.example.com:5060 | "example.com" 或 "hotelA.example.com"
多租户作用 | 不直接隔离 | 可以逻辑隔离不同租户或业务

<hr>
<h2>4. <strong>举个酒店场景例子</strong></h2>
<p>假设你有一台 SIP 服务器，IP 是 <code inline="">203.0.113.10:5060</code>，上面跑着 FreeSWITCH，接入两家酒店：</p>
<ul>
<li>
<p><strong>Hotel A</strong> → realm: <code inline="">hotelA.example.com</code></p>
</li>
<li>
<p><strong>Hotel B</strong> → realm: <code inline="">hotelB.example.com</code></p>
</li>
</ul>
<p>即使它们都连接同一个 <code inline="">203.0.113.10:5060</code>，<br>
但注册的时候：</p>
<ul>
<li>
<p>Hotel A 设备：</p>
<pre><code>REGISTER sip:hotelA.example.com SIP/2.0
</code></pre>
</li>
<li>
<p>Hotel B 设备：</p>
<pre><code>REGISTER sip:hotelB.example.com SIP/2.0
</code></pre>
</li>
</ul>
<p>认证时，服务器根据 realm 判断它是哪个酒店的账号，然后限制呼叫只在本 realm 内互通，这样避免了“跨酒店乱打电话”。</p>
<hr>
<p>✅ 总结一句：</p>
<ul>
<li>
<p><strong>SIP 服务器（5060）</strong> 是电话交换机的“门牌号”。</p>
</li>
<li>
<p><strong>Realm</strong> 是认证时的“房间号”，用来划分和隔离不同的用户组。</p>
</li>
</ul>
<hr>
<p>如果你愿意，我可以帮你画一个 <strong>SIP realm 与服务器关系图</strong>，让你直观看到两者的区别和交互过程，这样酒店多租户的场景就更容易理解了。<br>
你要我画吗？</p></body></html>
</body>
</html>

@jimforr

jimforr, 2025-08-15:
@liangtea 你好，我大概理解你的需求了，以下两点满足其一即可
1.User可以支持@#符号以让服务器区分客户端属于哪个租户
2.realm可自定义配置

针对第一点chatgpt的回复：
在 SIP URI 中，username 部分通常 不能包含 @ 字符，因为 @ 是 SIP URI 的语法中用来分隔 用户标识符（username） 和 主机部分（host/domain） 的关键字符。
在 SIP URI 中，username 部分通常 不能包含 # 字符，因为 # 字符在 URI 中有特殊含义。具体来说，# 是 URI 的 分隔符，用于指示 URI 中的 fragment（片段标识符），并且其后面的内容通常是用来指定资源的特定部分或某些附加参数。

针对第二点再确认下，因为第一条注册消息不会携带realm字段，是希望收到401回复之后的第二条注册消息realm字段使用自己配置的吗？如是，我会尽快提供可修改realm的方法。

[image]

liangtea, 2025-08-15:
针对第二点再确认下，因为第一条注册消息不会携带realm字段，是希望收到401回复之后的第二条注册消息realm字段使用自己配置的吗？如是，我会尽快提供可修改realm的方法。

是的。是这个意思。麻烦你了。

liangtea, 2025-08-20:
这是我freeswitch的目录结构（/usr/local/freeswitch/conf/directory）：
├── hotelA.com
│   ├── 1000.xml
│   ├── 1001.xml
│   ├── 1002.xml
│   ├── brian.xml
│   ├── default.xml
│   ├── example.com.xml
│   └── skinny-example.xml
├── hotelA.com.xml
├── hotelB.com
│   ├── 1000.xml
│   ├── 1001.xml
│   ├── 1002.xml
│   ├── brian.xml
│   ├── default.xml
│   ├── example.com.xml
│   └── skinny-example.xml
└── hotelB.com.xml

然后在以下目录配置相应配置就可以实现多域区分不同酒店了。
root@test:/usr/local/freeswitch/conf# find . -name "*.xml" -exec grep -H "hotelA"  {} \;
./directory/hotelA.com.xml:  <domain name="hotelA.com">
./directory/hotelA.com.xml:      <group name="hotelA.com">
./directory/hotelA.com.xml:       <X-PRE-PROCESS cmd="include" data="hotelA.com/*.xml"/>
./dialplan/hotelA.com.xml:  <context name="hotelA.com">
./dialplan/hotelA.com.xml:    <X-PRE-PROCESS cmd="include" data="hotelA.com/*.xml"/>
@jimforr 以上是我的freeswitch的配置。配置了两个域。你是否可以用这种配置来测试一下？

jimforr, 2025-08-20:
根据域名区分同名用户是SIP服务器的功能，客户端分别配置url如下是否可满足需求？
tcp://1001:123456@[hotelA.com:5060](hotelA.com:5060)
tcp://1001:123456@[hotelB.com:5060](hotelB.com:5060)

jimforr, 2025-08-25:
@liangtea 请稍等，我正在尝试配置不同域来测试

jimforr, 2025-08-29:
> [@jimforr](https://github.com/jimforr) 谢谢回复。不过我查了下这个rtc header应该是设置不了的。可能是我描述问题，客户端应该不能叫realm，一般叫域。我们是按照你VOIP的示例用freeswitch作为服务器的，那么你那边是否可以实现这样一个非常常见的需求：两个酒店，分别叫酒店A和酒店B，A和B都有101，两个不同酒店的ESP分别注册101时，怎么告诉服务器它属于哪个酒店呢？不然服务器如何判断它是哪个酒店呢？经过我们测试，现有VOIP代码无法实现这个需求。但是Linphone（比较出名的SIP客户端，有PC端手机端），市面上的SIP VOIP电话机都能实现这个需求。比如Linphone，我的用户名填[101@hotelA.com.cn](mailto:101@hotelA.com.cn)，连接到freeswitch，那么[101@hotelA.com.cn](mailto:101@hotelA.com.cn)只可以和[102@hotelA.com.cn](mailto:102@hotelA.com.cn)互通，同理[101@hotelB.com.cn](mailto:101@hotelB.com.cn)只可以和[102@hotelB.com.cn](mailto:102@hotelB.com.cn)互通（注意：这里的hotelA.com.cn并不等于SIP服务器的域名或者IP，虽然他们可以一样，一样的时候就是不区分不同酒店了，就是不允许有两个101了）。

@liangtea 你好，我尝试使用linphone按照你说的配置用户名，但是无法添加账户，日志显示这个用户名组装的sip uri是个bad uri，如图所示。
[image]

在 SIP URI 中，username 部分通常 不能包含 @ 字符，因为 @ 是 SIP URI 的语法中用来分隔 用户标识符（username） 和 主机部分（host/domain） 的关键字符。
所以请问你是如何使用linphone等客户端按照你说的方式成功设置用户名的

learnin9, 2025-09-07:
这块儿问题我来总结下吧，我也遇到这个情况了。当注册时，是采用这种注册到域，然后有个proxy服务器指向，这个是实际的IP地址+port方式。
如果需要的话，我这边是可以有实际的测试环境，直接通过microsip进行注册测试的。这种其实就是多租户的方式实现的隔离。

learnin9, 2025-09-16:
我贴下信令内容吧，如下是microsip的注册信令内容,其中106.xx.xx.xx代表着是服务器实际IP
注册截图：

[image]

**客户端发送SIP注册**
```
2025/09/16 14:15:51.224388 222.92.111.50:56467 -> 172.19.11.189:50600
REGISTER sip:106.xx.xx.xx:50600 SIP/2.0
Via: SIP/2.0/UDP 192.168.4.10:56467;rport;branch=z9hG4bKPj90c0a2512f8140d18aa1bbc6db5a5b1a
Route: <sip:106.xx.xx.xx:50600;lr>
Max-Forwards: 70
From: "8001" <sip:8001@nj.yilelink.com>;tag=2e10089051ef408288f3642a0236f525
To: "8001" <sip:8001@nj.yilelink.com>
Call-ID: de816951098f4d998aca32340b598fe6
CSeq: 32040 REGISTER
User-Agent: MicroSIP/3.21.5
Contact: "8001" <sip:8001@192.168.4.10:56467;ob>
Expires: 300
Allow: PRACK, INVITE, ACK, BYE, CANCEL, UPDATE, INFO, SUBSCRIBE, NOTIFY, REFER, MESSAGE, OPTIONS
Content-Length:  0
```

**服务器返回401鉴权挑战**
```
2025/09/16 14:15:51.224981 172.19.11.189:50600 -> 222.92.111.50:56467
SIP/2.0 401 Unauthorized
Via: SIP/2.0/UDP 192.168.4.10:56467;rport=56467;branch=z9hG4bKPj90c0a2512f8140d18aa1bbc6db5a5b1a;received=222.92.111.50
From: "8001" <sip:8001@nj.yilelink.com>;tag=2e10089051ef408288f3642a0236f525
To: "8001" <sip:8001@nj.yilelink.com>;tag=SNree2F6va1ga
Call-ID: de816951098f4d998aca32340b598fe6
CSeq: 32040 REGISTER
User-Agent: FreeSWITCH-mod_sofia/1.10.12-release~64bit
Allow: INVITE, ACK, BYE, CANCEL, OPTIONS, MESSAGE, INFO, UPDATE, REGISTER, REFER, NOTIFY, PUBLISH, SUBSCRIBE
Supported: timer, path, replaces
WWW-Authenticate: Digest realm="nj.yilelink.com", nonce="d03b2cf5-d3f3-41e9-b214-0aacbceb53f9", algorithm=MD5, qop="auth"
Content-Length: 0
```

**客户端发送注册带鉴权**
```
REGISTER sip:106.xx.xx.xx:50600 SIP/2.0
Via: SIP/2.0/UDP 192.168.4.10:56467;rport;branch=z9hG4bKPjed5ec79020e04cd5aa6913cc73166516
Route: <sip:106.xx.xx.xx:50600;lr>
Max-Forwards: 70
From: "8001" <sip:8001@nj.yilelink.com>;tag=2e10089051ef408288f3642a0236f525
To: "8001" <sip:8001@nj.yilelink.com>
Call-ID: de816951098f4d998aca32340b598fe6
CSeq: 32041 REGISTER
User-Agent: MicroSIP/3.21.5
Contact: "8001" <sip:8001@192.168.4.10:56467;ob>
Expires: 300
Allow: PRACK, INVITE, ACK, BYE, CANCEL, UPDATE, INFO, SUBSCRIBE, NOTIFY, REFER, MESSAGE, OPTIONS
Authorization: Digest username="8001", realm="nj.yilelink.com", nonce="d03b2cf5-d3f3-41e9-b214-0aacbceb53f9", uri="sip:106.xx.xx.xx:50600", response="ebd90c4be24e4386ac82f7e5bfb45e72", algorithm=M
D5, cnonce="655f39b84c88477ca4728cee23cb4264", qop=auth, nc=00000001
Content-Length:  0
```
接下来就是注册成功的消息了，我就不发送了。
@jimforr 您可以根据这里面的信令看下您的SIP协议栈是否有修改的地方

liangtea, 2025-09-16:
@learnin9 看起来你比我专业。能否给@jimforr 提供几个测试账号？

jimforr, 2025-09-25:
@liangtea 非常抱歉，由于服务器问题耽搁了很长时间，我添加了一个domain配置，用来实现配置域的功能，在示例里按照如下添加即可
    esp_rtc_config_t sip_service_config = {
        .uri = uri,
        .local_addr = _get_network_ip(),
        .acodec_type = RTC_ACODEC_G711A,
        .data_cb = &data_cb,
        .event_handler = _esp_sip_event_handler,
        .domain = "hotelA.com",
    };

发送的报文如图，服务器会把realm更改为发起注册时填充的域名，且可以注册成功，应该可以满足你的需求

[image]

稍后我会提供更新后的lib库

liangtea, 2025-09-25:
@jimforr 非常感谢。辛苦了，期待尽快更新库，谢谢😁

jimforr, 2025-09-26:
> [@jimforr](https://github.com/jimforr) 非常感谢。辛苦了，期待尽快更新库，谢谢😁

@liangtea 已更新在上方回复中了

另外，@learnin9 你提供的报文起了非常大的帮助，感谢！

liangtea, 2025-09-26:
问题已经解决。谢谢 @jimforr

closed 2025-09-26 by liangtea