espressif/esp-adf #1517: new i2c issue with  esp_codec_dev 1.4.0

aiotfactory, 2025-08-12:
Hello,

The codes work with old i2c driver, but after upgrading to new i2c driver  and the following components, it reports error.
idf is 5.4.1

## IDF Component Manager Manifest File
dependencies:
  espressif/button:
    version: 3.4.0
  espressif/esp-sr: 2.1.4
  espressif/esp32-camera: 2.1.2
  
  esp_codec_dev:
    version: "==1.4.0"
    public: true

  espressif/esp_websocket_client: ">=1.5.0"

I (4330) AFE: AFE Version: (1MIC_V250121)
I (4332) AFE: Input PCM Config: total 2 channels(1 microphone, 1 playback), sample rate:16000
I (4341) AFE: AFE Pipeline: [input] -> |AEC(SR_HIGH_PERF)| -> )| -> |VAD(WebRTC)| -> |WakeNet(wn9_xiaomeitongxue_tts,)| -> [output]
I (4352) app_sr: load wakenet:wn9_xiaomeitongxue_tts
I (4357) app_sr: audio_chunksize=512, feed_channel=3
I (4362) app_sr: ------------detect start------------

I (4362) app_sr: audio_tts_task start
E (4362) i2c.master: I2C transaction unexpected nack detected
I (4370) app_sr: afe_chunksize: 512
E (4375) i2c.master: s_i2c_synchronous_transaction(924): I2C transaction failed
I (4370) app_sr: app_stt_task start
E (4386) i2c.master: i2c_master_multi_buffer_transmit(1186): I2C transaction failed
I (4391) ES8311: Work in Slave mode
E (4396) I2C_If: Fail to write to dev 30
I (4403) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (4408) ES8311: Work in Slave mode
I (4408) I2S_IF: STD Mode 1 bits:16/16 channel:2 sample_rate:16000 mask:1
E (4414) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet
I (4426) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (4426) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
I (4431) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:1
I (4436) I2S_IF: STD Mode 0 bits:16/16 channel:2 sample_rate:16000 mask:1
E (4443) i2s_common: i2s_channel_disable(1191): the channel has not been enabled yet
I (4457) I2S_IF: channel mode 0 bits:16/16 channel:2 mask:1
E (4462) i2s_std: i2s_channel_reconfig_std_slot(331): invalid state, I2S should be disabled before reconfiguring the slot
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x42016d74  PS      : 0x00060c30  A0      : 0x82016fe7  A1      : 0x3fcd5eb0
--- 0x42016d74: set_drv_fs at D:/project/datacollection/Espressif2/projects/dc02/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:140

A2      : 0x3c186618  A3      : 0x00000001  A4      : 0x00000010  A5      : 0x3c186d48
A6      : 0x00000001  A7      : 0x00000010  A8      : 0x00000000  A9      : 0x3fcd5e80
A10     : 0x00000103  A11     : 0x3fcd5ee8  A12     : 0x3c1295c8  A13     : 0x00001169
A14     : 0x3c129518  A15     : 0x3c186618  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
--- 0x400570e8: memset in ROM
--- 0x400570f3: memset in ROM

Backtrace: 0x42016d71:0x3fcd5eb0 0x42016fe4:0x3fcd5f60 0x42017032:0x3fcd5fa0 0x420171fd:0x3fcd5fd0 0x420164da:0x3fcd6000 0x4200fb91:0x3fcd6020 0x403858aa:0x3fcd6060
--- 0x42016d71: set_drv_fs at D:/project/datacollection/Espressif2/projects/dc02/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:140
--- 0x42016fe4: set_fs at D:/project/datacollection/Espressif2/projects/dc02/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:263
--- 0x42017032: check_fs_compatible at D:/project/datacollection/Espressif2/projects/dc02/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:277
--- 0x420171fd: _i2s_data_set_fmt at D:/project/datacollection/Espressif2/projects/dc02/managed_components/espressif__esp_codec_dev/platform/audio_codec_data_i2s.c:452
--- 0x420164da: esp_codec_dev_open at D:/project/datacollection/Espressif2/projects/dc02/managed_components/espressif__esp_codec_dev/esp_codec_dev.c:179
--- 0x4200fb91: mic_read_task at D:/project/datacollection/Espressif2/projects/dc02/main/app_audio_record.c:432
--- 0x403858aa: vPortTaskWrapper at D:/project/datacollection/Espressif2/frameworks/esp-idf-v5.4.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

TempoTian, 2025-08-13:
I see your are using ES7210, could you please try to use test APP to verify it works or not.
There is a glitch count which needs to be set or else it will show the NACK error, if you are not using it, recommend to add it and have a try.
```c
    i2c_master_bus_config_t i2c_bus_config = {0};
    i2c_bus_config.clk_source = I2C_CLK_SRC_DEFAULT;
    i2c_bus_config.i2c_port = port;
    i2c_bus_config.scl_io_num = TEST_BOARD_I2C_SCL_PIN;
    i2c_bus_config.sda_io_num = TEST_BOARD_I2C_SDA_PIN;
    i2c_bus_config.glitch_ignore_cnt = 7; // Need set this value to 7
    i2c_bus_config.flags.enable_internal_pullup = true;
    return i2c_new_master_bus(&i2c_bus_config, &i2c_bus_handle);
```

aiotfactory, 2025-08-14:
> Does the test app also NG? You can also use i2c_tools https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools to check whether can find device or not. By the way, are you using DEV board or customized board? We are working on DEV board use new i2c driver for long time, and have not found issue yet. For es8311, the first time write to reg will get NACK, afterwards it will success.

thanks alot, I used customized board, and btw, is there any way to be success for the first time for 8311?

TempoTian, 2025-08-15:
Code is already commited in https://github.com/espressif/esp-adf/commit/45d3c3edc2a2d0d136bf1b0d15ad8d27f28143c6
A register can improve es8311 i2c noise tolerance, but it it already merged.

Licyxx, 2025-12-29:
Hello author, I also encountered the same problem during my recent development. Could you tell me if it has been solved and how you solved it?