espressif/esp-adf #1518: Dependency on esp_wifi_remote / esp_hosted adds a few hundred k to image size for ESP32-P4

ftab, 2025-08-12:
## Environment

- Audio development kit:      custom board
- [Required] Module or chip used:  ESP32-P4
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): v5.5
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): v2.7-135-g10dc80cd
- Build system:         idf.py
- [Required] Running log:         n/a
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it): riscv32-esp-elf-gcc (crosstool-NG esp-14.2.0_20241119) 14.2.0
- Operating system:     all
- Using an IDE?: no
- Power supply:         On board

## Problem Description

Without adding esp_hosted and esp_wifi_remote as dependencies to the project, it fails to build:

```
In file included from /Users/mac/esp/esp-adf/components/esp_peripherals/periph_wifi.c:30:
/Users/mac/esp/esp-adf/components/esp_peripherals/periph_wifi.c: In function '_wifi_init':
/Users/mac/esp/esp-adf/esp-idf/components/esp_wifi/include/esp_wifi.h:311:26: error: 'CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM' undeclared (first use in this function); did you mean 'CONFIG_ETH_DMA_RX_BUFFER_NUM'?
  311 |     .static_rx_buf_num = CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM,\
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Everything depends on esp_peripherals. I've tried to weasel my way out of including any network connectivity in this project because it doesn't need it. But periph_wifi.c persists.

### Expected Behavior

There should be no dependency on wifi in projects that do not need it

### Actual Behavior

I need to add esp_hosted and esp_wifi_remote to my project, and it adds extra size to my binary

### Steps to Reproduce

Build a project that doesn't need WiFi, like pipeline_sd_card_mp3_control

Add esp_hosted and esp_wifi_remote as dependencies to get it to build

See idf.py size-components: libespressif__esp_hosted.a, etc.

ftab, 2025-08-14:
Tried that, now got this:

```
In file included from /Users/mac/esp/esp-adf/components/wifi_service/include/wifi_ssid_manager.h:31,
                 from /Users/mac/esp/esp-adf/components/wifi_service/src/wifi_service.c:37:
/Users/mac/esp/esp-adf/components/wifi_service/src/wifi_service.c: In function 'wifi_sta_setup':
/Users/mac/esp/esp-adf/esp-idf/components/esp_wifi/include/esp_wifi.h:311:26: error: 'CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM' undeclared (first use in this function); did you mean 'CONFIG_ETH_DMA_RX_BUFFER_NUM'?
  311 |     .static_rx_buf_num = CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM,\
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

audio_stream requires esp_actions requires wifi_service

ftab, 2025-08-28:
> +if (CONFIG_SOC_WIFI_SUPPORTED OR CONFIG_ESP_HOSTED_ENABLED)
> +    list(APPEND COMPONENT_PRIV_REQUIRES wifi_service dueros_service)
> +endif()

Just tried to build for a target that does have WiFi and found out I can't do this (https://github.com/espressif/esp-idf/issues/16002#issuecomment-2897503836)

So I can't have it both ways. I either have to have an ESP-ADF that has wifi_service and dueros_service explicitly listed in PRIV_REQUIRES, or it is explicitly excluded. Can't have it both ways depending on Kconfig. Hmmmm...have to wait for IDF 6.0 I guess, or figure out some way to do this by IDF_TARGET only...which won't work with P4 projects that have hosted wireless chips.

TempoTian, 2025-08-29:
I have commit code to CI build pipeline use previous mentioned patch, it build success for all examples. The patch not build wifi_service source code at all, instead of change component dependency module CMakeLists.txt on it. I think it can meet your request, if not use wifi on P4. I see your attached build error log, I think you may missing add following options.
```
+if (CONFIG_SOC_WIFI_SUPPORTED OR CONFIG_ESP_HOSTED_ENABLED)
 list(APPEND COMPONENT_SRCS src/esp_wifi_setting.c
                     src/wifi_service.c
                     src/wifi_ssid_manager.c
                     smart_config/smart_config.c
                     blufi_config/blufi_config.c
                     blufi_config/blufi_security.c)
+endif()
```

closed 2025-09-09 by espressif-bot