espressif/esp-adf #1521: 使用自定义开发板运行 COZE_WS_APP 例程报错 init_coze_chat failed

yuangezhizao, 2025-08-16:
## Environment

- Audio development kit: 立创·实战派ESP32-C3开发板
- Audio kit version (for ESP32-LyraT/ESP32-LyraT-Mini/ESP32-S3-Korvo-2): -
- [Required] Module or chip used: ESP32-C3
- [Required] IDF version (run ``git describe --tags`` in $IDF_PATH folder to find it): v5.5.0
- [Required] ADF version (run ``git describe --tags`` in $ADF_PATH folder to find it): master
- Build system: idf.py
- [Required] Running log: -
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` in your project folder to find it): -
- Operating system: macOS
- (Windows only) Environment type: -
- Using an IDE?: VSCode
- Power supply: USB

## Problem Description

### Expected Behavior
-

### Actual Behavior
[image]

### Steps to Reproduce
1. 根据文档修改了相关配置：https://github.com/espressif/esp-adf/blob/master/examples/ai_agent/coze_ws_app/README_CN.md
2. 运行例程报错 Failed to create coze_data_pull_task，不知道应该从哪儿开始排查？

P.S. 发现代码中是调用[libcoze_websocket.a](https://github.com/espressif/esp-adf/blob/master/components/esp_coze/libs/esp32c3/libcoze_websocket.a)，想问下没有公开源码的原因，是为了优化运行效率吗？

shootao, 2025-08-21:
@yuangezhizao 你这个是不是没有开 psram

shootao, 2025-08-22:
c系列是没有psram 的， 最近有将 coze 中的 两个 task 去掉， 等 release 了 Q 你一下

yuangezhizao, 2025-08-24:
硬件换成刚复刻的[sparkbot-45coll](https://oshwhub.com/45coll/sparkbot-45coll)的头部后，除了有些`i2c`初始化的报错外，终于跑通例程了！😁

[image]

[image]

[image]

P.S. 资源点消耗的速度是真快 2333 😂

[image]

yuangezhizao, 2025-08-25:
才发现我新增的`LCKFB_C3_DEV`原来在`tempotian__codec_board`里已经存在，只不过名字是`XD_AIOT_C3`

这个`XD`迷惑性的前缀实在是没反应过来

[image]

Xiejz959, 2025-08-27:
两位大佬好，我也是跟楼主一样的日志报错。但我用的板子是esp32 lyrat mini v1.2。我有两个问题想咨询一下。  
一个是我再menuconfig里面选了板子的型号，但是型号只到v1.然后board_config.h里面并没有lyrat mini这个型号的相关配置，是需要自己配吗?  
第二就是任务创建失败的问题。我查了相关文档，也问了gpt，初步察觉是板子闪存不够的问题，这有没有解决方案？ 
PS：新手小白刚刚上手这个板子，目前就是想跑通这个例程，达到一个智能对话的效果，有没有好的解决方案？  
抱歉借楼问这个问题，如有打扰属实抱歉

NicLam26, 2025-09-25:
> c系列是没有psram 的， 最近有将 coze 中的 两个 task 去掉， 等 release 了 Q 你一下

> [@yuangezhizao](https://github.com/yuangezhizao) 你这个是不是没有开 psram

@shootao 请问这个release是在0.6.0版本里面吗？