espressif/esp-adf #1529: CZE_WS_APP例子报错

sourxsunny, 2025-09-07:
I (10166) ESP_COZE_CHAT: WEBSOCKET_EVENT_CONNECTED
I (10166) ESP_COZE_CHAT: WS connected
E (10167) HTTP_CLIENT: esp_http_client_init(753): Failed to convert addr type 8
I (10168) ESP_COZE_CHAT: Log ID: 021757228087918fdbddc02002702980000000000000040a274d7
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

**websocket连上后，会报Failed to convert addr type 8**

I (833) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (840) spi_flash: detected chip: gd
I (843) spi_flash: flash io: qio
W (846) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (858) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (864) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (871) main_task: Started on CPU0
I (883) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (883) main_task: Calling app_main()
E (908) COZE_CHAT_WS: Failed to get SPIFFS partition information (ESP_ERR_INVALID_STATE)
I (909) example_connect: Start example_connect.
I (910) pp: pp rom version: e7ae62f
I (913) net80211: net80211 rom version: e7ae62f
I (918) wifi:wifi driver task: 3fcbce7c, prio:23, stack:6656, core=0
I (932) wifi:wifi firmware version: cc7319b
I (932) wifi:wifi certification version: v7.0
I (933) wifi:config NVS flash: enabled
I (934) wifi:config nano formatting: disabled
I (939) wifi:Init data frame dynamic rx buffer num: 32
I (943) wifi:Init static rx mgmt buffer num: 5
I (947) wifi:Init management short buffer num: 32
I (952) wifi:Init dynamic tx buffer num: 32
I (956) wifi:Init static tx FG buffer num: 2
I (960) wifi:Init static rx buffer size: 1600
I (964) wifi:Init static rx buffer num: 10
I (968) wifi:Init dynamic rx buffer num: 32
I (972) wifi_init: rx ba win: 6
I (974) wifi_init: accept mbox: 6
I (978) wifi_init: tcpip mbox: 32
I (981) wifi_init: udp mbox: 6
I (983) wifi_init: tcp mbox: 6
I (986) wifi_init: tcp tx win: 5760
I (989) wifi_init: tcp rx win: 5760
I (993) wifi_init: tcp mss: 1440
I (995) wifi_init: WiFi IRAM OP enabled
I (999) wifi_init: WiFi RX IRAM OP enabled
I (1003) phy_init: phy_version 701,f4f1da3a,Mar  3 2025,15:50:10
I (1041) wifi:mode : sta (7c:df:a1:fd:dc:ec)
I (1041) wifi:enable tsf
I (1043) example_connect: Connecting to Redmi Note 9 Pro...
W (1043) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I (1051) example_connect: Waiting for IP(s)
I (3527) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (3527) wifi:state: init -> auth (0xb0)
I (3559) wifi:state: auth -> assoc (0x0)
I (3592) wifi:state: assoc -> run (0x10)
I (3695) wifi:connected with Redmi Note 9 Pro, aid = 31, channel 1, BW20, bssid = 6e:e7:b4:5d:54:72
I (3695) wifi:security: WPA2-PSK, phy: bgn, rssi: -38
I (3698) wifi:pm start, type: 1

I (3701) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3709) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (3717) wifi:dp: 2, bi: 102400, li: 4, scale listen interval from 307200 us to 409600 us
I (3725) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (4909) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:7edf:a1ff:fefd:dcec, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4949) esp_netif_handlers: example_netif_sta ip: 192.168.221.193, mask: 255.255.255.0, gw: 192.168.221.214
I (4949) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.221.193
I (4956) example_common: Connected to example_netif_sta
I (4961) example_common: - IPv4 address: 192.168.221.193,
I (4966) example_common: - IPv6 address: fe80:0000:0000:0000:7edf:a1ff:fefd:dcec, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4976) adc_button: calibration scheme version is Curve Fitting
I (4982) adc_button: Calibration Success
I (4985) button: IoT Button Version: 4.1.3
I (4989) ESP_COZE_CHAT: wss_url: wss://ws.coze.cn/v1/chat?bot_id=7544659009606500378
I (4997) ESP_COZE_CHAT: token: Bearer pat_ChZmRpzW1IEKupktA4NVnUHKQnI0XqdSNHapbUb9tvnza7IA1E88Ufsdffdfwefwe
I (5007) ESP_COZE_CHAT: WEBSOCKET_EVENT_BEGIN
I (5011) websocket_client: Started
I (5112) wifi:<ba-add>idx:0 (ifx:0, 6e:e7:b4:5d:54:72), tid:0, ssn:3, winSize:64
I (7909) example_connect: Got IPv6 event: Interface "example_netif_sta" address: 240e:047c:0cb0:4c89:7edf:a1ff:fefd:dcec, type: ESP_IP6_ADDR_IS_GLOBAL
I (10166) ESP_COZE_CHAT: WEBSOCKET_EVENT_CONNECTED
I (10166) ESP_COZE_CHAT: WS connected
E (10167) HTTP_CLIENT: esp_http_client_init(753): Failed to convert addr type 8
I (10168) ESP_COZE_CHAT: Log ID: 021757228087918fdbddc02002702980000000000000040a274d7
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x421547f7  PS      : 0x00060230  A0      : 0x820111be  A1      : 0x3fcadea0  
--- 0x421547f7: esp_http_client_set_method at E:/esp32_v5.4.0/src/v5.4/esp-idf/components/esp_http_client/esp_http_client.c:1195
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000094  A5      : 0x00000008
A6      : 0x3fccc410  A7      : 0x3fcaded0  A8      : 0x820d1817  A9      : 0x3fcade60
A10     : 0x00000000  A11     : 0x3c17a4f8  A12     : 0x3c17a8f4  A13     : 0x000027b7
A14     : 0x3c17a4f8  A15     : 0x3fccc3f4  SAR     : 0x0000001a  EXCCAUSE: 0x0000001d
EXCVADDR: 0x0000006c  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
--- 0x400570e8: memset in ROM
--- 0x400570f3: memset in ROM

Backtrace: 0x421547f4:0x3fcadea0 0x420111bb:0x3fcadec0 0x42010fb5:0x3fcadf90 0x4200d4ad:0x3fcae060 0x4200d23b:0x3fcae110 0x4215bbaf:0x3fcae160
--- 0x421547f4: esp_http_client_set_method at E:/esp32_v5.4.0/src/v5.4/esp-idf/components/esp_http_client/esp_http_client.c:1194
--- 0x420111bb: http_client_post at ??:?
--- 0x42010fb5: esp_coze_chat_start at ??:?
--- 0x4200d4ad: init_coze_chat at E:/esp32_v5.4.0/coze_ws_app/main/coze_chat_app.c:81
--- (inlined by) coze_chat_app_init at E:/esp32_v5.4.0/coze_ws_app/main/coze_chat_app.c:214
--- 0x4200d23b: app_main at E:/esp32_v5.4.0/coze_ws_app/main/main.c:71
--- 0x4215bbaf: main_task at E:/esp32_v5.4.0/src/v5.4/esp-idf/components/freertos/app_startup.c:208

ELF file SHA256: 043dd898a

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379f14
--- 0x40379f14: esp_restart_noos at E:/esp32_v5.4.0/src/v5.4/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:162
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x16d0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xec8
load:0x403cb700,len:0x3160
entry 0x403c8950
I (31) boot: ESP-IDF v5.4.2-568-g11edebdbbe-dirty 2nd stage bootloader
I (31) boot: compile time Sep  7 2025 10:07:56
I (31) boot: Multicore bootloader
I (33) boot: chip revision: v0.1
I (36) boot: efuse block revision: v1.2

shootao, 2025-09-09:
@jimforr @sourxsunny 现在的版本不会有这个问题了，因为这块代码是开出来的

jingyuan2017, 2025-10-17:
> [@jimforr](https://github.com/jimforr) [@sourxsunny](https://github.com/sourxsunny) 现在的版本不会有这个问题了，因为这块代码是开出来的

请问这个怎么解决？试了0.6.0也是这样，idf版本5.4.2